MakeraInc/MakeraCAM #10: Pad clearing path leaves a little notch of solder mask

shaun-leach, 2024-04-27:
[image]
[image]

Not a huge deal but it would be nice if the clearing path didn't do that

shaun-leach, 2024-04-27:
[image]

Interestingly the CAM software shows the paths as being closed.

leighost24, 2024-04-28:
Could you provide the Gerber files and MKC project file? We will track and analyze the cause of the problem.

shaun-leach, 2024-04-28:
[ContinuitySMT-F_Mask.zip](https://github.com/MakeraInc/MakeraCAM/files/15143806/ContinuitySMT-F_Mask.zip)
Here you go!

leighost24, 2024-07-04:
Fixed in V0.1.2.0

closed 2024-07-04 by leighost24