MakeraInc/MakeraCAM #18: Adding manual tabs automatically generates one in corner

beardedamish, 2024-06-07:
When going through a process of adding tabs, if the appropriate pixel isn't clicked on the profile, if it doesn't put the tab where I've clicked, it either disengages the profile or it automatically creates a tab in the upper righthand corner.  You must clear all tabs to fix it.  Undo last tab or removable tabs would be desired.

dpevnifb, 2024-06-28:
This is actually a big issue (i spend on avg 30-40 min redoing this part till it works and sometime after 4-5 attempts the whole project crash and i have to redo everything...) is there any progress on this? (on the slack it was mentioned that it will be adressed in the upcomming version, but i see it still not under work)

leighost24, 2024-07-04:
In version V0.1.2.0, we have optimized the contour tab and added a feature for timed automatic saving and recovery of files.

closed 2024-07-04 by leighost24