MakeraInc/MakeraCAM #21: Cannot add Drill file generated by EasyEda

renarj, 2024-07-24:
I am trying to get a small test PCB imported, the Top layer import works fine and the pads show up fine, however the Drill holes do not get imported. When i add the DRL file, it allows me to add it but nothing happens.

Attached the Gerber package that was generated by EasyEda:
[Gerber_PCB13_2024-07-24.zip](https://github.com/user-attachments/files/16357789/Gerber_PCB13_2024-07-24.zip)

Contents of the Drill file:
`
;TYPE=PLATED
;Layer: PTH_Through
;EasyEDA Pro Client v2.1.63.af010a1c.225d47, 2024-07-24 07:57:11
;Gerber Generator version 0.3
M48
METRIC,LZ,000.00000
;Hole size 1 = 0.30480 METRIC
T01C0.30480
;Hole size 2 = 1.90002 METRIC
T02C1.90002
%
G05
G90
T01
X01393102Y02097851
X01493102Y02097851
X01393102Y01997851
X01493102Y01997851
T02
X02095500Y00546100
X02095500Y01181100
X02921000Y02425700
X03556000Y02425700
M30
`

leighost24, 2024-07-24:
I tested it with the latest version 0.1.4.0. There was no problem with the parsing of the DRL&GDD files. Please try the following solutions:
1. Please update to the latest version and try again.
2. Hide other layers in the Layers Tree, leaving only Drill-related layers visible, and check if they are displayed correctly.
3. Import only DRL & GDD files and check whether they are normal. 

[image]

renarj, 2024-07-24:
I just tried this without importing any other layers and still the same issue i am afraid. I am on version "MakeraCAM Version  0.1.4.0  (Beta)". I am running on MacOS, is there a Log file or anything i can look at why it is then not opening on my system?

leighost24, 2024-07-24:
You are right, the Mac version does have this issue, we will fix it in the next version.