MakeraInc/MakeraCAM #22: add Linux support

petep0p, 2024-09-25:
please consider adding a Linux version of this software. thank you!

manuel-imboden, 2024-12-07:
+1 @MakeraInc Are there currently any plans to support Linux in the future?

ss32, 2025-02-09:
+1 anything but windows

bernieke, 2025-02-27:
FYI, it seems to install and start fine on wine. Haven't tried a full workflow yet though.

markqvist, 2025-02-27:
Linux support is what we need! Would really appreciate this!

p1um1oco, 2025-02-28:
Agreed, linux support would be great, I am trying to eliminate Windows from my life and only have two programs left to go.  Those two run inside a windows VM.

jeffc, 2025-02-28:
> FYI, it seems to install and start fine on wine. Haven't tried a full workflow yet though.

I got it running under Lutris, which is just Wine plus some free GPU configuration and tweaks, without any issues (or, at least, without any issues that I don't also see on Windows 😢 )

markqvist, 2025-02-28:
Thanks @jeffc, that's good to know!

Still, a native Linux build would really be preferable. A lot of people in the target demographic of machines like this use Linux for their entire design/production toolchain, and having to switch over to a Windows machine/VM is a major productivity and creativity killer.

manuel-imboden, 2025-03-04:
@markqvist Hit the nail on the head there – 100% agreed.
I'd love to hear @MakeraInc 's take on this.

ss32, 2025-03-04:
> I'd love to hear @MakeraInc 's take on this.

Per what I was told by a mod on Discord there is no plan to support Linux.  The app is built on QT so I don't know what's stopping them from supporting Linux other than motivation.

markqvist, 2025-03-04:
Even providing a Linux build with the disclaimer that it is "community supported" would be a big plus for a lot of people I think.

If it builds for both Windows and macOS currently, there's not likely to be any real *technical* issues with providing a Linux build, at least.

Even just a raw `x86_64` binary build, and I'm sure community members would be happy to help bundling it to various distribution formats such as AppImage, etc. Put it out, wait about 8.5 seconds, and it's on AUR ;)

rvalle, 2025-04-18:
No linux support means no real maker product, sorry.

petep0p, 2025-04-20:
So while it _does_ run under wine, at least on my install (generic wine under Ubuntu 24.04), i consistently get a fatal crash when trying to import an image. So far, everything else seems to be working, though....