MakeraInc/MakeraCAM #24: Windows app fails to load in W10

ss32, 2025-02-09:
I installed the app, typed in my email and then the app crashes.  Reinstalling didn't help.  The logo pops up and then nothing happens. I've tried launching from the command prompt but there isn't any helpful output.

bernieke, 2025-02-23:
Same here, on a freshly installed Windows 10.

bernieke, 2025-02-24:
Same thing on a freshly installed, fully updated, Windows 11.

What's going on Makera?

leighost24, 2025-02-26:
According to previous analysis, it may be a problem with the graphics card driver version. It is recommended to update the graphics card driver to the latest public version driver. If there is a separate graphics card, it is recommended to switch to the independent graphics card first.

ss32, 2025-03-04:
This was solved by running in a VM with GPU passthrough.  Apparently the reliance on a GPU renders the application unusable in a VM unless it has access to a discrete GPU.

closed 2025-03-04 by ss32

bernieke, 2025-03-04:
I was using qemu/kvm with a nvidia gpu, which unfortunately doesn't yet support pass-through.

I solved the problem for myself by running makera cam using wine instead, which seems to work just fine.