MakeraInc/MakeraCAM #26: Makera CAM 2.0.0 None of the open icons work

clschone, 2025-02-27:
macOS sequoia
Cannot open any file or save anything 
Permissions issue for files and folders?

iamahumanbeing, 2025-02-27:
Yes, I have the same problem as well. MacOS Sequoia 15.4. I did not have this problem in the previous version.

naldoxable, 2025-02-27:
Same here, Beta of 15.4 (24E5206s).

Also i noticed since the previous version that if i drag the window to a side of the screen for the auto resize, it crashes immediately with a segmentation fault. Maybe i should open another issue for that.

clschone, 2025-02-27:
To be clear, this is not an Open app issue. I already resolved that with Privacy and Security. Software open and I can go into either other first options. But anything that requires folder/disk access do not work. Can’t even save a dummy test file.

leighost24, 2025-03-05:
We have conducted verification on the MacBook Air M3 version 15.3.1 and were unable to replicate the issue that was reported. 

You can execute the following command in the terminal and then check to see if the issue persists: xattr -rc /Applications/MakeraCAM.app.

The next Mac version of MakeraCAM will include a digital signature, and we recommend verifying the issue again once it has been released.

clschone, 2025-03-05:
Thanks for the feedback. That command did have a slight affect.
I can now save an empty file. So that’s progress. the Open functions still do not work, and I cannot drag and drop a file, any kind, into MakerCam.
But that save function never popped up before that command

lawrencesuen, 2025-03-06:
Still happening even after applying the xattr command (This is on MacOS 15.4)

I know for some Mac apps, when permission is needed to access user folders, usually an OS dialog pops up requesting for permission. Perhaps this app needs to request this permission.

leighost24, 2025-03-06:
> Thanks for the feedback. That command did have a slight affect. I can now save an empty file. So that’s progress. the Open functions still do not work, and I cannot drag and drop a file, any kind, into MakerCam. But that save function never popped up before that command

Currently, the app likely does not support dragging and dropping files to open them directly. You need to use the program's built-in functions to load model files or MKC project files. I wonder if this operation method is causing the issue.

clschone, 2025-03-26:
Not sure if this is helpful but this is all I can get out of the console.
default	20:31:15.546133-0400	MakeraCAM	order window front conditionally: 24e related: 0
default	20:31:15.554018-0400	MakeraCAM	[0x60000299c000] activating connection: mach=false listener=false peer=false name=com.apple.TextInputUI.xpc.CursorUIViewService
default	20:31:15.554445-0400	MakeraCAM	[0x60000299c000] invalidated after the last release of the connection object
default	20:31:15.559376-0400	kernel	Sandbox: ContextStoreAgent(695) allow file-read-data /Applications/MakeraCAM.app
default	20:31:15.559664-0400	kernel	Sandbox: ContextStoreAgent(695) allow file-read-data /Applications/MakeraCAM.app/Contents
default	20:31:15.559889-0400	kernel	Sandbox: ContextStoreAgent(695) allow file-read-data /Applications/MakeraCAM.app/Contents/Info.plist
default	20:31:15.560393-0400	kernel	Sandbox: ContextStoreAgent(695) allow file-read-data /Applications/MakeraCAM.app
default	20:31:15.560905-0400	kernel	Sandbox: ContextStoreAgent(695) allow file-read-data /Applications/MakeraCAM.app/Contents/MacOS/MakeraCAM
default	20:31:15.561443-0400	kernel	1 duplicate report for Sandbox: ContextStoreAgent(695) allow file-read-data /Applications/MakeraCAM.app/Contents/MacOS/MakeraCAM
default	20:31:15.561451-0400	kernel	Sandbox: ContextStoreAgent(695) allow file-read-xattr /Applications/MakeraCAM.app/Contents/MacOS/MakeraCAM
default	20:31:15.571216-0400	MakeraCAM	[0x7ff698a05400] activating connection: mach=false listener=false peer=false name=(anonymous)
default	20:31:15.571324-0400	MakeraCAM	[0x7ff67a522ba0] activating connection: mach=false listener=false peer=false name=(anonymous)
default	20:31:15.588462-0400	MakeraCAM	-[TUINSCursorUIController activate:]_block_invoke: Create CursorUIViewService: TUINSRemoteViewController
default	20:31:15.588655-0400	MakeraCAM	[0x7ff67a1bcf10] invalidated because the current process cancelled the connection by calling xpc_connection_cancel()
default	20:31:15.588684-0400	MakeraCAM	[0x7ff6b879e8b0] invalidated because the current process cancelled the connection by calling xpc_connection_cancel()
error	20:31:15.588812-0400	MakeraCAM	ViewBridge to RemoteViewService Terminated: Error Domain=com.apple.ViewBridge Code=18 UserInfo={com.apple.ViewBridge.error.hint=this process disconnected remote view controller -- benign unless unexpected, com.apple.ViewBridge.error.description=NSViewBridgeErrorCanceled}
default	20:31:15.711817-0400	MakeraCAM	order window: 253 op: 1 relative: 24e related: 0
default	20:31:15.718848-0400	MakeraCAM	order window: 24e op: 0 relative: 0 related: 0
default	20:31:15.806720-0400	MakeraCAM	[0x600002982fd0] activating connection: mach=false listener=false peer=false name=com.apple.MTLCompilerService
default	20:31:15.852605-0400	MTLCompilerService	Compilation BEGIN (ParentProcessName=MakeraCAM_4335) Build request: MTLBuildFunctions - pipeline
default	20:31:15.862458-0400	MTLCompilerService	Compilation SUCCESS (ParentProcessName=MakeraCAM_4335) (TYPE=MTLBuildFunctions - pipeline): Time 9.837125 ms
default	20:31:15.863090-0400	MTLCompilerService	Compilation BEGIN (ParentProcessName=MakeraCAM_4335) Build request: MTLBuildFunctions - pipeline
default	20:31:15.866796-0400	MTLCompilerService	Compilation SUCCESS (ParentProcessName=MakeraCAM_4335) (TYPE=MTLBuildFunctions - pipeline): Time 3.687041 ms
default	20:31:15.921792-0400	MakeraCAM	order window front conditionally: 24d related: 0
default	20:31:15.992628-0400	MakeraCAM	[0x7ff6b9059f90] activating connection: mach=false listener=false peer=false name=(anonymous)
default	20:31:15.994750-0400	MakeraCAM	[0x6000029831b0] activating connection: mach=false listener=false peer=false name=com.apple.TextInputUI.xpc.CursorUIViewService
default	20:31:15.995174-0400	MakeraCAM	[0x7ff6b8793470] invalidated because the current process cancelled the connection by calling xpc_connection_cancel()
default	20:31:15.996507-0400	MakeraCAM	[0x6000029831b0] invalidated after the last release of the connection object
default	20:31:15.999080-0400	MakeraCAM	order window: 253 op: 0 relative: 0 related: 0
default	20:31:16.001483-0400	MakeraCAM	[0x7ff669011380] activating connection: mach=false listener=false peer=false name=(anonymous)
default	20:31:16.001564-0400	MakeraCAM	[0x7ff698b1c210] activating connection: mach=false listener=false peer=false name=(anonymous)
default	20:31:16.062004-0400	MakeraCAM	-[TUINSCursorUIController activate:]_block_invoke: Create CursorUIViewService: TUINSRemoteViewController
default	20:31:16.062515-0400	MakeraCAM	[0x7ff67a522ba0] invalidated because the current process cancelled the connection by calling xpc_connection_cancel()
default	20:31:16.062546-0400	MakeraCAM	[0x7ff698a05400] invalidated because the current process cancelled the connection by calling xpc_connection_cancel()
default	20:31:16.095006-0400	MakeraCAM	order window: 255 op: 1 relative: 24c related: 0
default	20:31:16.095141-0400	MakeraCAM	order window: 24c op: 0 relative: 0 related: 0
default	20:31:16.249222-0400	MakeraCAM	order window: 255 op: 0 relative: 0 related: 0
default	20:31:17.769923-0400	MakeraCAM	order window front conditionally: 256 related: 0
default	20:31:18.442013-0400	MakeraCAM	order window: 257 op: 1 relative: 256 related: 0
default	20:31:18.442152-0400	MakeraCAM	order window: 256 op: 0 relative: 0 related: 0
default	20:31:18.585475-0400	MakeraCAM	order window: 257 op: 0 relative: 0 related: 0
[... cut ...]
        BackgroundState = 2;
        BundleID = "com.apple.console";
        Hide = 0;
        Path = "/System/Applications/Utilities/Console.app";
    },
        {
        BackgroundState = 2;
        BundleID = "";
        Hide = 0;
        Path = "/Applications/MakeraCAM.app";
    }
)
default	20:37:35.709578-0400	mds_stores	[<private>] makeReadOnly <private>
default	20:37:35.768940-0400	mds_stores	[<private>] makeReadOnly <private>
default	20:37:35.780119-0400	mds_stores	[<private>] makeReadOnly <private>

clschone, 2025-04-02:
Same issue with the new 2.1.0 release on a macOS sequoia 15.4. No access to the file directory

ojarva, 2025-04-13:
I have the same issue with MacOS 15.4, including with the latest version of MakeraCAM. There's no longer need to run `xattr` with digital signatures, but neither opening any files or saving anything works - dialogs do not open at all.

M-Unimaster, 2025-04-14:
@omanikhi found a [workaround](https://github.com/MakeraInc/MakeraCAM/issues/43#issuecomment-2800105361) and posted it in the other issue: