MakeraInc/MakeraCAM #27: [MakeraCam 2.0.0] Rotary Relief Doesn't Recognize STEP files as 3D Models

jeffc, 2025-02-28:
When I import a STEP file as a 3D object and try to create a rotary relief toolpath for it, I get the error "no 3D model selected".  The same object generates a toolpath with the other 3D operations (relief, contour, etc) without complaint, even in the 4axis workspace.

jeffc, 2025-02-28:
I should also note that the transformation widgets for move, rotate, and scale don't appear when they should for a STEP file either, though manually entering the scale does work

leighost24, 2025-03-31:
> When I import a STEP file as a 3D object and try to create a rotary relief toolpath for it, I get the error "no 3D model selected". The same object generates a toolpath with the other 3D operations (relief, contour, etc) without complaint, even in the 4axis workspace.

This feature will be supported in the next version.

The issue about transformation widgets may take some time to fix it.