MakeraInc/MakeraCAM #29: Problems with the 2D pocket tool in versions 1.5 and 2.0

lucsch, 2025-03-03:
In version 1.5, if several tool sizes are entered, the 2D pocket tool will start with the first and continue with the second, as illustrated below. 

Note : some objects seem to be ignored

[image]

In version 2.0, if several tool sizes are entered, the 2D pocket tool only seems to take the first one into account. 

[image]

Finally, in both versions, if only one tool is selected, then all objects are correctly processed. 

[image]

In the case of designs with large and small objects (typically PCBs), it would be faster and more interesting to be able to use several tools to remove large surfaces and then deal with the details.

The sample file is attached bellow

[image]

jmzimmerle, 2025-03-08:
Thank you! I thought it was me :) . I have noticed the same issue in 2.0 where the software seems to ignore anything but the first tool in the list.

hoskinsw, 2025-03-26:
Same issue, just commenting to bring more awareness to it. Maybe it worked in an earlier version? I will need to go look at versions before 1.5

leighost24, 2025-03-31:
We have noticed this issue and will fix it in the next version.