MakeraInc/MakeraCAM #30: Zooming with the wheel is unusable (MacOS)

lucsch, 2025-03-03:
In MakeraCAM, when zooming with the mouse wheel, the zoom factor is so large that the drawing immediately disappears from the screen. It is therefore impossible to zoom in or out with the mouse. 

Mouse settings are set to minimum.

https://github.com/user-attachments/assets/7709fded-53c4-46aa-a68d-9cbd605b3e87

As a workaround, would it be possible to have shortcut keys for zoom functions (+ and - for example). It would also be nice to have shortcuts for changing tools (selection, rotation, pan, zoom).

System: MacOS 15.3.1 Sequoia

David-Large, 2025-03-03:
Seconding this request — zooming with the mouse at any setting is practically unusable.

leighost24, 2025-03-05:
We would like to confirm the specific model of mouse you are using. Based on our internal testing, mechanical mouse wheels do not exhibit uncontrollable zoom behavior. The reported issue is currently only observed with Mac trackpad operations.

David-Large, 2025-03-05:
I should have clarified, sorry — yes, I'm using a Mac trackpad. Good to know that a mechanical mouse wheel will work!