MakeraInc/MakeraCAM #32: Cannot Add Tabs

SoundsSerious, 2025-03-09:
Perhaps similar to #18, however I have an issue where on an outside contour operation the "add tabs" context doesn't add tabs to the selected outline. Dont think screen shots will help here, its hard to show something not happening.

I watched the tutorial video but i dont believe I'm doing anything different. I am using v2.0.0 which is the default from the site download but says its a beta release on github. 

I will retry with v1.5.

amanenk, 2025-03-09:
Hello. I am facing the same issue on [v0.2.0.0](https://github.com/MakeraInc/MakeraCAM/releases/tag/v0.2.0.0) version. I have installed [v0.1.5.0](https://github.com/MakeraInc/MakeraCAM/releases/tag/v0.1.5.0) and tabs feature works.

amanenk, 2025-03-09:
I found a workaround on makera discord. If you reduce resolution of the screen to 1080p tab add funciton might start working.

SoundsSerious, 2025-03-09:
I'll give that a shot. My screen resolution is 3440x1440 for debugging purposes.

leighost24, 2025-03-31:
This issue will be fixed in the next version.