MakeraInc/MakeraCAM #36: Change Email

SoundsSerious, 2025-03-15:
hit the trial expiration today, however I guess the email i was using wasn't recognized, however I forgot which one i was using.

What can I do to reset the email? 

By the way are you using kivy for this CAM tool and/or the carvera controller?

SoundsSerious, 2025-03-15:
Ok think I found a solution on windows to resetting the license, go to `C:\Users\<your_user>\AppData\Roaming\MakeraCAM` and look for the `license.dat` file where you can find the email used to register. Delete the file and re-register on initialization.

For @MakeraInc I think you might to take a look at the way `Remarkable` does things where each device is given a one time device key after setting up in their customer portal. 

That being said, the email I used to register was the one I purchased my Carvera with so seems like something is broken upstream.

SoundsSerious, 2025-03-15:
Hmm while this lets you re-add an email It seems to remember the old one, so looks like I am waiting on `info@makera.com` like everyone else :)

leighost24, 2025-03-31:
There were some issues with the previous authentication logic, but it should have been resolved now. The reset operation can be changed through the menu item [Help - Log out] to change the user.

closed 2025-03-31 by leighost24