MakeraInc/MakeraCAM #46: Display Scale below 100% for high DPI Displays

unrat, 2025-04-13:
I use a 27" 4k monitor and the UI elements and texts are ridiculously large. This also artificially reduces the space for the design area.

As there is already a setting to scale the display up to 200% would it also be possible to add options to scale below 100%? Please add a 75% and maybe even a 50% selection as well.

It would also be desirable to have a finer gradation of scale adjustment.

BR

MakeraCAM Version  0.2.1.0  (Beta)

leighost24, 2025-04-14:
What is the resolution of your device? What is the magnification set by the system? If Windows OS has set a magnification greater than 100%, please set the magnification in MakeraCAM to the magnification of Windows OS.

unrat, 2025-04-15:
Native resolution of the Monitor is 3840x2160
Magnification in Windows is set to 150%.

Setting the scale in MakeraCAM to 150% does not change the size of UI or texts, it is exactly the same size as in automatic scale setting.

Maybe I was a bit unclear about the problem I have with the UI. I am using a 4k Monitor because with the higher resolution one can scale down UI elements in contrast to a full hd display because you have more pixels per UI element and therefore a much sharper appereance. Even when you would scale down the UI by 25 or so percent you still end up with more pixels than a full hd display would have for the same elements. Simply said the UI elements can be much smaller but still are better visoble than with an full hd display od the same size factor.

unrat, 2025-08-06:
Found a workaround for my case.

Deactivate the dpi scaling for the MakerCAM via registry:

1. open registry editor
2. go to `Computer\HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\SideBySide`
3. add a DWORD value named `PreferExternalManifest`
4. set its value to '1'

Create manifest-file in the installation folder of MakeraCAM.exe with following content:

`<?xml version="1.0" encoding="UTF-8" standalone="yes"?>`
`<assembly xmlns="urn:schemas-microsoft-com:asm.v1" manifestVersion="1.0">`
`  <application xmlns="urn:schemas-microsoft-com:asm.v3">`
`    <windowsSettings>`
`      <dpiAware xmlns="http://schemas.microsoft.com/SMI/2005/WindowsSettings">True</dpiAware>`
`    </windowsSettings>`
`  </application>`
`</assembly>`

Set display scale in MakeraCAM itself to 175% or 200%, as the scaling is swapped. The larger the scaling is set in MakeraCAM the smaller it actually will be displayed