MakeraInc/MakeraCAM #51: 3d relief tool path won't directly generate

petep0p, 2025-04-30:
i "think" this happened after the last update: 
i don't seem to be able to generate a 3D relief tool path any longer on the latest Windows version. i was able to create and mill one earlier, but i wanted to make a finer fine path to get better detail and clean it up. now the tool path just generates an overlay at the very top of the stock boundary with absolutely no variation in Z.

please let me know what further information it would be best to provide

leighost24, 2025-04-30:
Please provide the operating system & version, CPU model, and whether or not you have a discrete graphics card. Also can you provide us with the mkc file so we can analyze the amount of data in the model.

petep0p, 2025-05-01:
OS: 
- Edition	Windows 11 Pro
- Version	23H2
- OS build	22631.5189

Processor:
- AMD Ryzen 3 5400U with Radeon Graphics

Graphics (onboard GPU):
- AMD Radeon Graphics

regarding the mkc file - i'm not able to upload the full archive here, as it's not a supported file type. it's 72MB in size, though. I have attached the "makera.mkc" file found within, in the hopes that this is the file you meant. (added ".txt" on the end to allow upload)

[makera.mkc.txt](https://github.com/user-attachments/files/19996694/makera.mkc.txt)

petep0p, 2025-05-04:
I uploaded the project file to gDrive. [Here's a link to it.](https://drive.google.com/file/d/1wCmCFRGaUOJOiKpeYOvSrKPxldXUk9Hj/view?usp=drive_link)

petep0p, 2025-05-10:
any update on this issue? I'm still unable to generate a second fine relief toolpath. It always now generates as a sweep on top with no Z variation.

leighost24, 2025-05-13:
We verified this MKC project, the calculations for each toolpath are correct. From the results you provided for each toolpath in the MKC program, the 0.2mm toolpath calculation is correct, the problem in the 0.1mm toolpath calculation, which will have double the amount of data than the 0.2mm toolpath data, it may be a limitation imposed by the performance of your device video memory or graphics card. 

We are also optimizing this part of the performance at this stage, to facilitate better adaptation to more hardware configurations.

petep0p, 2025-05-16:
thanks for getting back. here's the issue though: the 0.2mm tool path was generated using the previous version of makeraCAM. the 0.1mm was generated using the current version. if i attempt to generate any 2nd tool path using the current version (ie, even on a new project), it results in the same issue with no Z variation. so i don't think it's a hardware issue because it worked before