MakeraInc/MakeraCAM #64: Why are you forcing us to use the latest version !

FraserElectronics, 2025-08-04:
[image]

It is not acceptable to force us to use the latest version while MakerCAM is so buggy.

leighost24, 2025-08-05:
The discontinuation of certain historical versions is due to critical bugs in these versions that could lead to noticeable tool collisions or significant calculation errors. While the latest version has addressed many issues, there may still be some shortcomings, and we sincerely apologize for any inconvenience caused. MakeraCAM is still in its growth phase, and we are continuously working to optimize algorithms, improve stability, and make the software more robust and intelligent. We greatly appreciate your ongoing support and understanding, your feedback is invaluable to us.

FraserElectronics, 2025-08-05:
Thank you for your comments.