MakeraInc/MakeraCAM #75: Milling PCB

FraserElectronics, 2025-09-18:
I am having problems with MakeraCAM making PCBs. I started following Jason's video on PCBs but when I select my bottom copper layer and choose 2D Pocket - in Jason's video, it calls it Vector Pocket ! I get the following:

[image]

You can see than instead of creating isolation routing around the track and pads, the tool actually cuts along the track and pads !

What am I doing wrong?

Andy

oscer959, 2025-11-28:
Your board outline consist of two lines. It can be seen in your attached picture, but if you zoom in it will be even more clear. That is the reason for this behavior. Delete one of the two board outlines or just zoom in and deselect one of the two lines (shift - left click). With just the graphics for the traces and a single outline selected the pocket operation will work as intended. (If PCB uses e.g. ground plane deselect the board outline completely (or select both if dual lines)).

FraserElectronics, 2025-11-28:
@oscer959 

Thanks for that - I had not spotted the second board outline.
As you suggest, it works fine when you remove one of them.

BR

Andy

closed 2025-11-28 by FraserElectronics